ros-drivers/camera1394 #47: Image Flip / Mirror (FFMV USB)

mhkabir, 2015-02-06:
Hi, 

Having the ability to flip or mirror a image from a camera would be very useful (I'm using a FireFly MV on the USB bus) and our hardware mounting is such that I need one camera flipped horizontally.

According to PointGrey's Register Reference, it can be enabled by setting the register IMAGE_DATA_FORMAT(0x1048) , bit 23 to 1 or 0 as required. It will do image flip on-camera in DMA buffers.
https://pixhawk.ethz.ch/_media/dev/electronics/cameras/point_grey_digital_camera_register_reference.pdf

If this change isn't something that can be adapted to mainstream, I'm willing to do it myself, but would like some help on where and how to implement it best. Otherwise, if this can be implemented in software in the driver itself, that'd be awesome too, and would benefit all camera types.

Thanks,
Kabir

jack-oquin, 2015-02-07:
The current master branch of the driver source allows external ROS nodes to get and set register values via new ``GetCameraRegisters`` and ``SetCameraRegisters`` service calls. 

This new feature is not yet documented or released.

Would you please build the driver from source and test whether that interface works for your application?

jack-oquin, 2015-02-07:
Flipping the image can be done in software, but is probably better done outside camera1394, or any other specific camera driver.

If you want to pursue that approach, I would recommend writing a generic "image flipper" nodelet that can do the job for any image stream. That could be added to ``image_common`` or released as a separate package.

mhkabir, 2015-02-07:
Yep. I'm happy to do both these approaches and report back ;) A generic image_flipper nodelet would indeed be very useful. 

Thanks for the help!!

mhkabir, 2015-02-07:
@jack-oquin The offset referred to, how'd we use it to access a specific register? Can you give me a simple example please.

mhkabir, 2015-02-07:
@jack-oquin I think I've mostly got it. 

The value[]  that the Get service call returns is a bitmask,if I'm correct. So I should be able to modify bit 23 by - getting the value from the register, making value[23] whatever I desire, and writing the value back?

jack-oquin, 2015-02-07:
I didn't write that code, and it is still not documented. That is the main reason why it has not yet been released.

I think you have the general idea. The Get returns the contents of some particular register, and the Set changes its value. This feature was mostly introduced to support various non-standard Point Grey interfaces without adding non-standard code to the camera1394 driver.

Maybe if you figure out how to use it, you can help me document it. I don't own any of those cameras, making it hard for me to test.