ros-drivers/camera1394 #48: 100% CPU usage issues

mhkabir, 2015-05-21:
The driver causes nearly 100% CPU usage under some conditions:
* The camera is unplugged while the driver is running
* External triggering is used

Camera : PointGrey FireFly MV
OS : Ubuntu 14.04.02 LTS
ROS version : Indigo

jack-oquin, 2015-05-26:
Thanks for the report. I will try to reproduce it.

mhkabir, 2015-05-27:
I'm using trigger parameters :
* external_trigger = true
* trigger_polarity = ACTIVE_LOW

mhkabir, 2015-05-27:
You should just be able to unplug the camera while the driver is running and monitor the system load in ```htop``` to easily reproduce. 
Also another issue, the driver cannot be shutdown cleanly. Ctrl+C results in a lot of corrupted linked-list errors before it's  eventually terminated. ( I will follow up on this later today,  after updating to latest master)

mhkabir, 2015-06-01:
@jack-oquin Ping ;)

mhkabir, 2015-06-12:
@jack-oquin Any updates on this? Can you repro? I'm getting sorta desperate...

mhkabir, 2015-06-17:
@jack-oquin ping.

mhkabir, 2015-06-18:
@jack-oquin This commit is causing it : https://github.com/ros-drivers/camera1394/commit/d279c2d3a18a5cb59cadc17c74ca03a44588f496

jack-oquin, 2015-06-18:
Thanks for figuring that out. 

I am still having trouble finding time for maintaining this package, but don't give up.