ros-drivers/camera1394 #51: Unable to change Frame rate of BackFly USB 3  camera

MuhammadUsman90, 2016-03-22:
$ rosrun camera1394 camera1394_node _guid:=00b09d0100e96e58
[ WARN] [1458642111.606965852]: Comparing 00b09d0100e96e58 to 00b09d0100e96e58
[ INFO] [1458642111.665301569]: Found camera with GUID 00b09d0100e96e58
[ INFO] [1458642111.665498984]: camera model: Point Grey Research Blackfly BFLY-U3-03S2C
[ WARN] [1458642111.667959759]: Bayer pattern [ (4)] is invalid.
libdc1394 error: Generic failure: in dc1394_format7_get_max_image_size (format7.c, line 249): Could not get max image sizes

libdc1394 error: Generic failure: in dc1394_format7_get_max_image_size (format7.c, line 249): Could not get max image sizes

libdc1394 error: Generic failure: in dc1394_format7_get_max_image_size (format7.c, line 249): Could not get max image sizes

libdc1394 error: Generic failure: in dc1394_format7_get_max_image_size (format7.c, line 249): Could not get max image sizes

[ INFO] [1458642111.749695479]: Format7 unit size: (4x2), position: (2x2)
[ INFO] [1458642111.750066935]: Format7 region size: (648x488), offset: (0, 0)
[ INFO] [1458642111.788503000]: [00b09d0100e96e58] opened: format7_mode0, 15 fps, 400 Mb/s
[ INFO] [1458642112.130803109]: using default calibration URL
[ INFO] [1458642112.131337272]: camera calibration URL: file:///home/odroid/.ros/camera_info/00b09d0100e96e58.yaml
^C*** Error in `/home/odroid/catkin_ws/devel/lib/camera1394/camera1394_node': corrupted double-linked list: 0x00123668 ***
Aborted

jack-oquin, 2016-03-22:
If I remember correctly, the ``frame_rate`` has no effect when Format7 is selected. That is not explained clearly in the documentation, which should be improved.

Does it work with your device in some other modes?

fabrizioschiano, 2018-03-23:
I have the same problem. Did someone solved this problem ?

Thanks.